peterhinch/micropython_ir #1: TypeError: can't convert str to int

Flaming-crossbow, 2021-01-20:
In advance i must warn you, i am a complete micropython beginner, and i would also like to thank you for making these packages.

Anyway I followed the instructions and gave the test a try: 

from ir_rx.acquire import test
test()

And got a reading:

Waiting for IR data...
000  8986
001  4484
002   575
003   585
004   579
005   582
006   578
007   583
008   578
009   583
010   641
011   517
012   580
013   583
014   579
015   583
016   577
017   583
018   578
019  1658
020   581
021  1661
022   579
023  1656
024   582
025  1660
026   578
027  1659
028   579
029  1662
030   577
031  1662
032   574
033  1665
034   576
035   585
036   577
037  1658
038   580
039  1663
040   576
041   583
042   576
043  1660
044   581
045   582
046   579
047   583
048   577
049   584
050   576
051  1659
052   581
053   582
054   577
055   585
056   585
057  1650
058   582
059   583
060   579
061  1661
062   629
063  1606
064   582
065  1662
066   579

NEC

After that i gave the pyboard example a try but i am using an ESP32 minikit so i removed the PYB and LED lines:

import time
from machine import Pin
#REMOVED from pyb import LED
from ir_rx.nec import NEC_8  # NEC remote, 8 bit addresses

#REMOVED red = LED(1)

def callback(data, addr, ctrl):
    if data < 0:  # NEC protocol sends repeat codes.
        print('Repeat code.')
    else:
        print('Data {:02x} Addr {:04x}'.format(data, addr))

ir = NEC_8(Pin('X3', Pin.IN), callback)
while True:
    time.sleep_ms(500)
    red.toggle()

And i got the following error:
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
TypeError: can't convert str to int

This is line 14:  
ir = NEC_8(Pin('X3', Pin.IN), callback)

Any idea on what is going wrong?

peterhinch, 2021-01-21:
It's hard to read your code as the formatting is lost. Please create a code block with a line starting ```python.

After the last line add an end line consisting of three backtick characters ```.

If you click the preview tab you should see you code properly formatted, like this:
```python
def foo():
    print('Hello')
```
The only obvious bug I can see is the line
```python
red.toggle()
```
but maybe I'll spot something else when I can see the formatting.

Flaming-crossbow, 2021-01-23:
Thanks for the tip!

```python

import time
from machine import Pin
#REMOVED from pyb import LED
from ir_rx.nec import NEC_16  # NEC remote, 8 bit addresses

#REMOVED red = LED(1)

def callback(data, addr, ctrl):
    if data < 0:  # NEC protocol sends repeat codes.
        print('Repeat code.')
    else:
        print('Data {:02x} Addr {:04x}'.format(data, addr))

ir = NEC_16(Pin('X3', Pin.IN), callback)
while True:
    time.sleep_ms(500)
    #red.toggle()

```
also without red.toggle() i get the same error.

peterhinch, 2021-01-23:
In all MicroPython code you have to specify pins, and the designation for them depends on the device in use.

The way to map buttons is first to use the code you have to determine the output of the buttons you intend to use. Write this output down, checking it is consistent if you press the button more than once. Then rewrite your callback to detect those values and do whatever you plan to do.

Flaming-crossbow, 2021-01-23:
Thanks again!

closed 2021-01-23 by Flaming-crossbow