peterhinch/micropython_ir #4: Error: Invalid start bit and overrun.

piyush1609, 2021-06-21:
Hello sir, hope you are doing well in this pandemic situation, and thank you for your library. 
My problem: 
I'm using Arduino UNO as a Transmitter that sends NEC data every 0.5 sec and using PICO (rp2040) as a receiver but problem is that most of the time my Pico gives an error like invalid start bit or overrun, I know what this error is but this should not occur...  
I'm facing this issue for the past 2-3 days, I'm using TSOP1838, and it's working perfectly fine in Arduino as a receiver, but I want to implement it on Pico.
[image]
 
I also tried to measure the raw NEC signal and I found that pico is missing some bits. I have copy-pasted values on a spreadsheet and compared them. 
[necdata(mpython).xlsx](https://github.com/peterhinch/micropython_ir/files/6689616/necdata.mpython.xlsx)
[image]

I don't know whether this is the right place to post a question since you are the author of this code, I posted here.
Thank you have a nice day :)

piyush1609, 2021-06-22:
My TV remote supports NEC protocol, I tried to send the signal it's the same problem most of the time it will show an invalid start pulse or overrun, and in 1 in 6 cases it will give the correct output, I even replaced TSOP but its the same issue...

peterhinch, 2021-06-23:
Thanks for testing with a TV remote. I don't really know how to help here. I'm puzzled by your spreadsheet. I'm sure you've figured out what follows, but I'll spell it out to make sure we're on the same page.

The spreadsheet shows a reasonably accurate start which should be 9ms followed by 4.5ms, but I can't make sense of the rest of the data. Every data bit should have a 560μs ON period followed by either 560μs or 1690μs OFF. Your data shows no obvious correlation with this, neither me nor my library can decode it. As for the reason for this, an oscilloscope would doubtless show you - I feel there must be an electrical problem but I don't know how to guess at this remotely.

One observation. Your start is typically 9257μs on, 4226μs off. This adds up to 13483μs. Official times are 9000+4500=13500μs. So your total time is highly accurate, but the on time is +257μs and the off time is -274μs. This seems to hint at a slow on-off transition.

piyush1609, 2021-06-24:
Okay, sir, I will look into that and let you know if I got any doubt, till then I would close this issue. 
Thank you :)

closed 2021-06-24 by piyush1609