peterhinch/micropython_ir #6: which gpio number of raspberry pico to be inserted(IR Receiver)

notjiangjiu, 2021-07-08:
I don't see in RECEIVER.md which GPIO interface raspberry pie needs to plug in，can you provide the gpio number？
[image]

notjiangjiu, 2021-07-08:
Hello, I didn't seem to notice. Later, I found a platform of RP2 in qcquire.py. Is this raspberry pie Pico? Its GPIO is set to 16

After testing, the result is data 16 addr 0000 Ctrl 00, no address. Is this normal?

peterhinch, 2021-07-08:
RP2 is the chip used in the Pico. In my testing with the Pico I used GPIO 16, but any pin can be used if the code is adapted to suit. Data received will depend on the remote control you're testing.

peterhinch, 2021-07-08:
I have updated [RECEIVER.md](https://github.com/peterhinch/micropython_ir/blob/master/RECEIVER.md) to clarify which pins `acquire.py` uses on the various platforms. Thank you for pointing this out.

closed 2021-07-08 by peterhinch

notjiangjiu, 2021-07-08:
I'm a novice in Python. Although you can't understand some of the code in your program, I feel that your code is very concise and the function call is so ingenious, which is incredible

In addition, due to my lack of ability, I need to use raspberry pie Pico to receive different remote control buttons and let Pico control the steering gear. Can you give me a hint of where to add such functions? Can you give me an example?

Thank you very much and I will learn from you

peterhinch, 2021-07-09:
If you look at [this sample](https://github.com/peterhinch/micropython_ir/blob/master/RECEIVER.md#3-the-driver) you just need to rewrite the callback: instead of just printing the received values, check for specific ones and trigger your actions. For example, say you want to turn a motor on and off:
```python
def callback(data, addr, ctrl):
    if data < 0:  # NEC protocol sends repeat codes.
        # You might want to handle repeat codes in some way
    else:
        if data == 3:  # Values determined by testing the remote with the original code
            motor.on()
        elif data == 4:
            motor.off()
```

notjiangjiu, 2021-07-09:
Thank you very much for your patience。