peterhinch/micropython_ir #12: Getting Error: Overrun while receiving data on ESP32

icheered, 2022-01-21:
Following the example code on an ESP32 I get an "Error: overrun" every time I receive data but I don't know what causes this or how to prevent this error from happening. I _am_ receiving data correctly confirmed using an oscilloscope. This issue happens using both NEC_16 and NEC_8. 

Receiver code
```python
import time

import machine
import gc
from ir_rx.nec import NEC_16
from ir_rx.print_error import print_error

def callback(data, addr, ctrl):
    print("Callback is called")
    if data < 0:  # NEC protocol sends repeat codes.
        print("Repeat code.")
    else:
        print("Data {:02x} Addr {:04x}".format(data, addr))

def test():
    ir = NEC_16(machine.Pin(32, machine.Pin.IN), callback)
    ir.error_function(print_error)
    try:
        i = 0
        while True:
            i += 1
            print("Running: " + str(i))
            time.sleep(5)
            gc.collect()
    except KeyboardInterrupt:
        ir.close()

test()

```

Transmitter code for completeness (Works as expected as confirmed using oscilloscope).
```python
import machine
from ir_tx.nec import NEC

button = 0
buttonPin = machine.Pin(26, machine.Pin.IN)

transmitter = NEC(machine.Pin(12, machine.Pin.OUT))

while 1:
    if buttonPin.value() and not button:
        button = 1
        print("Button went low")

    elif not buttonPin.value() and button:
        button = 0
        print("Button went high " + str(i))
        i += 1
        transmitter.transmit(1, i)

```

What exactly is causing this issue? And how can I prevent it from happening?

peterhinch, 2022-01-21:
Your code looks fine.

An overrun error occurs if the receiver [sees more than 68 edges](https://github.com/peterhinch/micropython_ir/blob/d10a35788161b8bca341bcb1803c63c74c64ce96/ir_rx/nec.py#L20) so my first port of call would be to count them on your scope. It's possible that your remote is non-compliant or perhaps there is an issue with your decoder chip.

Another test that can be done if you have two micropython platforms and an IR emitter is to set one up as transmitter and the other as receiver.

The NEC protocol is widely used. I tested with a number of remotes and never saw this problem.

Theagainmen, 2022-04-15:
I'm having the exact same problem.

Is it possible to have both the receiver and sender on the same device? For my application I want the user to be able to send data, but sometimes also receive data. That's what I'm testing right now too, so maybe that is the problem?

icheered, 2022-04-15:
@Theagainmen it is absolutely possible to have the sender and receiver on the same device (got it to work fairly reliably). I don't fully remember what the solution to this problem was but I do remember tinkering with the circuit before getting it to work. If you have the tools I would recommend hooking up an oscilloscope to check if the signal looks correct.

Theagainmen, 2022-04-15:
Thanks a lot for your quick response! Turns out I didn't have the ground properly connected...

It seems to receive the signal correct occasionally now! However, is it normal that it's spamming error messages all the time when there are no IR signals being sent?

```
Invalid start pulse (getting these the most)
Error: bad block (these sometimes)
```

icheered, 2022-04-15:
Great! Yeah those messages pop up intermittently because of IR from other sources, that's nothing to worry about.

I also just remembered that the previously mentioned error still pops up sometimes and my solution was just to send the information a few times consecutively (only takes a few ms anyways) to increase chance of correct delivery.