peterhinch/micropython_ir #15: IR Emitter using Raspberry PICO module

EagleZodiac007, 2022-05-17:
Hi Peter,
I haven't get a clear picture for to IR emitter using RPI PICO.
RPI PICO and IR LED are enough for generating the signal or we need any external circuit ?

peterhinch, 2022-05-17:
You need to use a circuit like those described [here](https://github.com/peterhinch/micropython_ir/blob/master/TRANSMITTER.md#11-pyboard-wiring). The IR diode needs to be driven with a relatively high current which exceeds the capability of the output pins of microcontrollers.

closed 2022-05-17 by peterhinch

jouellnyc, 2024-02-03:
I tried to build this https://github.com/peterhinch/micropython_ir/blob/master/TRANSMITTER.md#11-pyboard-wiring but failed. I saw this https://www.adafruit.com/product/157 and it works well to transmit. Point in fact I heard in the Adafruit video that lots' of people have struggled to build a  circuit to emit IR frames "just right".  I blame my rudimentary EE skills...:P and was happy to pay the $4 :)