peterhinch/micropython_ir #16: Transmitting RC5 encoded IR signal with RPI pico

EagleZodiac007, 2022-05-20:
In transmitter.md you have mentioned about two push buttons. Is that each of them is for sending different signal? How should we connect it?
I have a requirement of waking up the philips tv from standby, how can i achieve that?what is the value i supposed to transmit?
Sorry if it is a basic thing if i'm asking. I'm relatively new to this field.
Please let me know if you can help

peterhinch, 2022-05-20:
You need to make a receiver as per the docs. Press the "wake up" button on the remote. You will then know the code necessary for the transmitter.

EagleZodiac007, 2022-05-23:
Hi Peter,
I see that, b.append is the function where we add the signal we wanted to encode. My doubt is what is address and data in that function?If i want to send a key code like 0x006c should i update which variable addr or data?

Another doubt is about hardware of transmitter, Is IR LED should be powered with external voltage source or I can power up it by using connecting pico VSYS pin  output to IR LED is Anode>? is that sufficient to trigger the signal?

peterhinch, 2022-05-23:
Please see [the Wikipedia article on RC5](https://en.wikipedia.org/wiki/RC-5). Commands comprise an address and data.

I suggest you use one of the circuits in my doc. Microcontroller pins are not capable of supplying the current that the IR LED requires.

EagleZodiac007, 2022-05-28:
Hi Peter,
I have tried everything as you mentioned in the document. But the signal is generated in IR LED. 
I have checked the signal coming from gpio pin in Oscilloscope. I was expecting RC6 encoded signal but it's not generated.