peterhinch/micropython_ir #21: Error: bad block    while running ir_rx.test on Pico

BambooStrategic, 2022-10-30:
Hello,

I am starting with the basic Receiver Test on a RP Pico  in order to read the Tx codes on a Samsung remote.
When I run the from ir_rx.test import test  , I then selected test (8) as well as test(), and when it starts to run I receive the Error: bad block  - see video-  [New Recording - 10/30/2022, 12:55:30 AM](https://vimeo.com/765357030/341f444809)

I have tried a few different times with the same results.

I have been successful using an arduino project to read the Samsung, but I am not getting the codes in the format that your application requires. The codes I was able to get look like this:  On/Off:  E0E040BF    volume up: E0E0E01F  volume down: E0E0D02F
Does the ir_rx.test application break those codes down into the format that is required for the Transmitter to function?

Any suggestions are appreciated. I am not an expert at any of this, and learning every day through the process.

thank you,
Paul

peterhinch, 2022-10-30:
The addition of Samsung is quite recent (last update 28th Sept) so firstly make sure you have the latest code. However I only have one example of a Samsung remote. It works, as does that of @cyrilchristin who submitted the latest fix. There do seem to be different versions of the Samsung protocol, and it is possible that your remote uses a variant that I haven't yet encountered.

The 38KHz chip should be correct according to [my measurement](https://github.com/peterhinch/micropython_ir/blob/master/README.md#31-carrier-frequencies).

BambooStrategic, 2022-10-30:
I was successful at getting the values using the Arduino, now using your latest Transmitter code.  

Does this look correct?

from machine import Pin
from ir_tx.nec import NEC
import time

led_onboard = Pin(25, Pin.OUT)
nec = NEC(Pin(17, Pin.OUT, value = 0))
NEC.samsung=True
addr = 0x707
dat = 0x7
rpt = 0xF8070707

while True:
    nec.transmit(addr,dat)
    time.sleep(0.5)
    nec.transmit(addr,rpt)
    time.sleep(0.5)
    led_onboard.value(1)
    time.sleep(5)
    led_onboard.value(0)

peterhinch, 2022-10-31:
To display Python code, please prefix it with three ` characters, with the same at the end. Then indenting is preserved (click the preview tab to see the effect).

Your code looks OK except for
```python
rpt = 0xF8070707
```
The data value passed to `nec.transmit` must be an 8-bit value, as in the rest of your code. I'm not sure where you got this number from.

As a matter of interest where did address `0x707` come from? The reason I ask is that my remote happens to use that address and it worked fine with the receiver.

BambooStrategic, 2022-11-02:
I built a different circuit using an Arduino to read the transmitter.  I still want to do this on a Pico using python.
Here is the Arduino project link:

https://mschoeffler.com/2021/05/01/arduino-tutorial-ir-transmitter-and-ir-receiver-hx-m121-hx-53-ky-005-ky-022-keyes-iduino-open-smart/#:~:text=A%20typical%20example%20for%20IR,undetectable%20to%20the%20human%20eye.

Here are the results that I received using on/off , Volume up   Volume down  .  

Received something...
Protocol=NEC Address=0x707 Command=0xE6 Raw-Data=0x19E60707 32 bits LSB first

Received something...
Protocol=NEC Address=0x707 Command=0xE6 Repeat gap=47600us

Received something...
Protocol=NEC Address=0x707 Command=0x2 Raw-Data=0xFD020707 32 bits LSB first

Received something...

Received something...
Protocol=NEC Address=0x707 Command=0x2 Repeat gap=47600us

Received something...
Protocol=NEC Address=0x707 Command=0x7 Raw-Data=0xF8070707 32 bits LSB first

Received something...
Protocol=NEC Address=0x707 Command=0x7 Repeat gap=47650us

Received something...
Protocol=NEC Address=0x707 Command=0x7 Raw-Data=0xF8070707 32 bits LSB first

Received something...
Protocol=NEC Address=0x707 Command=0x7 Repeat gap=47600us

Received something...
Protocol=NEC Address=0x707 Command=0xB Raw-Data=0xF40B0707 32 bits LSB first

Received something...
Protocol=NEC Address=0x707 Command=0xB Repeat gap=47600us

Received something...
Protocol=NEC Address=0x707 Command=0xB Raw-Data=0xF40B0707 32 bits LSB first

Received something...
Protocol=NEC Address=0x707 Command=0xB Repeat gap=47600us

peterhinch, 2022-11-02:
Here is the outcome of testing my Samsung remote. I have annotated it to explain which buttons I'm pressing. In each case I held the button down long enough for multiple messages to be sent.
```python
>>> from ir_rx.test import test
Test for IR receiver. Run:
from ir_rx.test import test
test() for NEC 8 bit protocol,
test(1) for NEC 16 bit,
test(2) for Sony SIRC 12 bit,
test(3) for Sony SIRC 15 bit,
test(4) for Sony SIRC 20 bit,
test(5) for Philips RC-5 protocol,
test(6) for RC6 mode 0.
test(7) for Microsoft Vista MCE.
test(8) for Samsung.

Hit ctrl-c to stop, then ctrl-d to soft reset.
>>> test(8)
running
running
Data 0x02 Addr 0x0707 Ctrl 0x00  # ON/OFF
running
Data 0x02 Addr 0x0707 Ctrl 0x00
Data 0x02 Addr 0x0707 Ctrl 0x00
Data 0x02 Addr 0x0707 Ctrl 0x00
Data 0x02 Addr 0x0707 Ctrl 0x00
Data 0x02 Addr 0x0707 Ctrl 0x00
running
Data 0x07 Addr 0x0707 Ctrl 0x00  # VOLUME UP
Data 0x07 Addr 0x0707 Ctrl 0x00
Data 0x07 Addr 0x0707 Ctrl 0x00
Data 0x07 Addr 0x0707 Ctrl 0x00
running
Data 0x0b Addr 0x0707 Ctrl 0x00  # VOLUME DOWN
Data 0x0b Addr 0x0707 Ctrl 0x00
Data 0x0b Addr 0x0707 Ctrl 0x00
Data 0x0b Addr 0x0707 Ctrl 0x00
Data 0x0b Addr 0x0707 Ctrl 0x00
Data 0x0b Addr 0x0707 Ctrl 0x00
running
running
>>> 
```
It seems identical to yours except for the initial message you got on pressing ON/OFF (the second message matched mine). So I'm baffled as to why yours didn't work with this test script. Could there be an electrical issue?

BambooStrategic, 2022-11-02:
Thank you Peter.

I am traveling now, but will try to build your receiver again this weekend
and try again with that.

I have some new components arriving as well to make the Tx circuit as well,
so I am hopeful I can create pair of  Tx and Rx circuits and get the
expected results.

So the 0x00 is what I should use as the third value?

peterhinch, 2022-11-03:
> So the 0x00 is what I should use as the third value?

If you mean the `toggle` arg to `transmit`, it defaults to 0. It is unused by Samsung. Further, the Samsung protocol doesn't seem to use repeat codes. So if you want to steadily increase volume, this should work:
```python
from machine import Pin
from ir_tx.nec import NEC
import time

led_onboard = Pin(25, Pin.OUT)
nec = NEC(Pin(17, Pin.OUT, value = 0))
NEC.samsung=True
addr = 0x707
dat = 0x7

while True:
    nec.transmit(addr, dat)
    time.sleep(0.5)
    led_onboard.value(1)
    time.sleep(5)
    led_onboard.value(0)
 ```
With these time values, volume change will be very gradual - you might want to reduce the 5s delay.