peterhinch/micropython_ir #23: test.py assistance

siennamediatest, 2023-02-03:
Sorry for the basic questions:
I'm a nursing assistant in a long-term care home, and have been working on a media centre project for residents with dementia. Integral to this is building a simplified remote, as residents tv remotes have been overcomplicated. The remote I'm building is going to have only 2 buttons (power, channel), and a buzzer that can be remotely pinged so it's easier to find (people lose their remotes all the time).
Browsing through the code of this library has been very illuminating but confusing. However, I have been able to run the test.py program and run test() to send NEC codes through my pi pico w to control the media centre. I use the media centre to change keycaps based on the test program's signals. 
Using the existing test.py program code, how can I bypass the test selection and automatically run the test() remote for NEC protocol?

peterhinch, 2023-02-03:
You could do something like this:
```python
import uasyncio as asyncio
from primitives import WaitAny, ESwitch
from ir_tx.nec import NEC

nec = NEC(Pin(20, Pin.OUT, value=0))
btn1 = ESwitch(Pin(10, Pin.IN, Pin.PULL_UP))  # Pushbutton linked to Gnd
btn2 = ESwitch(Pin(11, Pin.IN, Pin.PULL_UP))
evt1 = btn1.close  # Button press events
evt2 = btn2.close

async def main():
    while True:
        evt1.clear()
        evt2.clear()
        evt = await WaitAny((evt1, evt2)).wait()  # Pause until a button is pressed
        if evt is evt1:
            nec.transmit(1, 0)  # Adapt for your needs
        elif evt is evt2:
            nec.transmit(1, 10)

asyncio.run(main()
```

siennamediatest, 2023-02-03:
that worked perfectly! thank you so much for your help

closed 2023-02-04 by peterhinch