peterhinch/micropython_ir #27: Controlling Gree ACs with Pico

MrFarePlay, 2023-09-12:
First of all thank you for the amazing work.

I've been playing with the library for a few days trying to emulate a YACIFBI Gree AC remote control. I'm using the test and Player sections to get and transmit the raw signal. I'm getting the raw signal from my remote, but transmitting to the AC doesn't seem to work (trying to turn it off for simplicity).

Any guidance would be greatly appreciate.

I'm connecting the IR led directly to a Pico pin with resistor if that matters.

I have a Flipper zero which im using to test and see if the IR led is actually transmitting. Saving the saved signal from the IR led to the flipper and playing directly to the AC has no result. Saving the signal from the remote directly to the flipper and playing to the AC does work.

Thanks in advance.

peterhinch, 2023-09-14:
What is the outcome when you run [the test scripts](https://github.com/peterhinch/micropython_ir/blob/master/RECEIVER.md#21-test-scripts) with the remote control?

bestemy, 2023-10-03:
@MrFarePlay 
Please contact me on Twitter or Discord. This is the only place I hope you will see the message :(