peterhinch/micropython_ir #33: acquire.py and test.py usage

erich74, 2023-11-07:
Hi Peter,

First off I would really like to thank you for this IR lib.
I am using an M5StickC Plus device flashed with micropython on which I have attached an IR Receiver.
What I would like to achieve is being able to acquire an IR Remote signal and then display its protocol , address, and code (and display this information on the M5Stick screen).

I would like to be able to discover the protocol and then retrieve its address/code.

I guess I need a combination of your acquire.py and test.py examples.

So far I have slightly modified/simplified your acquire.py file so that I can just print out on the console the name of the protocol.
It is working fine.
Later I would like to get the address/code using the proper/matching protocol class.

My question for now is: say that I have identified the IR Remote is sending a NEC protocol, what would be the best way to modify the acquire.py file so that I can tell which NEC protocol this is about, NEC_8 or NEC_16? I will indeed need this information in order to call the right protocol class (and I would also like the user to know if it is NEC_8 or NEC_16)... of course same would apply for SONY

Thanks a lot for your answer!
Eric

peterhinch, 2023-11-08:
If you read [this doc](https://www.sbprojects.net/knowledge/ir/nec.php) I think you'll agree that you can't unambiguously distinguish NEC_8 from NEC_16. NEC_8 is characterised by the address being transmitted followed by its complement. If you examine the 16 address bits and find it has such a format the source is almost certainly NEC_8. But it could be NEC_16 with a pathological address value...

erich74, 2023-11-08:
Thanks Peter for your answer. Does it mean I could always use the NEC_16 class and it would retrieve the address/code even though the IR remote is sending NEC_8 protocol?

peterhinch, 2023-11-09:
I'm pretty sure that is the case. The address would comprise the 8 bit address and its complement.

erich74, 2023-11-09:
ok thank you very much for your help, I will try this out and let you know (I actually own both NEC_8 and NEC_16 remote controls).

One last question: do you think it could work the same for SONY? I mean using SONY_20 class for all sony protocols (12, 15 and 20)?

peterhinch, 2023-11-09:
I can't really add anything to my words in [the doc](https://github.com/peterhinch/micropython_ir/blob/master/RECEIVER.md#sony-classes): `SONY_20` should work for all, but the timing for repeats may be a bit marginal.

erich74, 2023-11-09:
Sure I understand, I will also give it a try if I can put my hands on Sony IR remotes.
Thanks again

closed 2023-11-09 by erich74