peterhinch/micropython_ir #35: Trying to get Apple Remote A1156 working, but every test() results in device reboot. (wemos C3)

Pax-IT, 2024-01-02:
Hi,

I'm trying to get an old Apple Remote A1156 to work with my ESP32-C3.

I'm working with a Lolin Wemos ESP32-C3: https://www.wemos.cc/en/latest/c3/index.html
Micropython: MicroPython v1.22.0 on 2023-12-27; LOLIN_C3_MINI with ESP32-C3FH4

I tried: `from ir_rx.test import test`
And test() or test(1) result in crash and reboot.

I finally tried: `from ir_rx.acquire import test`

which gave me this output:

```
>>> 
MPY: soft reboot
MicroPython v1.22.0 on 2023-12-27; LOLIN_C3_MINI with ESP32-C3FH4
Type "help()" for more information.
>>> from ir_rx.acquire import test
>>> test()
Waiting for IR data...
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40047a74  RA      : 0x420024b2  SP      : 0x3fca7ea0  GP      : 0x3fc97c00
TP      : 0x3fc6e63c  T0      : 0x4203a93e  T1      : 0x00000000  T2      : 0xffffffff
S0/FP   : 0x0000000e  S1      : 0x3fcddf5c  A0      : 0x00000000  A1      : 0x31da05cc
A2      : 0x0000000c  A3      : 0x00001e00  A4      : 0x00000008  A5      : 0x600c2000
A6      : 0x00010000  A7      : 0x00000000  S2      : 0x00001000  S3      : 0x00000000
S4      : 0x0000000c  S5      : 0x00000020  S6      : 0x3fca7ec8  S7      : 0x3fcddf4c
S8      : 0x74737973  S9      : 0x00000020  S10     : 0x00000001  S11     : 0x00000000
T3      : 0x00000052  T4      : 0x00000fa0  T5      : 0x0000013a  T6      : 0x00000002
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x31da05cc
MHARTID : 0x00000000

Stack memory:
3fca7ea0: 0x00000020 0x000000c8 0x00000000 0x00000000 0x3fcaa394 0x3fcaa370 0xffc1ffff 0x7fffffff
3fca7ec0: 0x00000000 0x00000003 0x003e0000 0x00000000 0x00000000 0x00000001 0x00000002 0x00000000
3fca7ee0: 0x3fca7f3c 0x3c14187c 0x00000004 0x3c141000 0x3c141000 0x000004c7 0x3fcaa150 0x3fcaaf66
3fca7f00: 0x3fffffff 0x3ffffffe 0x40000000 0x3fca7f64 0x80000000 0x00000001 0x3fcaa370 0x42006b0c
3fca7f20: 0x00000020 0x009354f4 0x00010000 0x00000000 0x3fcaa3a0 0x00000064 0x000003e8 0x00000006
3fca7f40: 0x3fffffff 0x3ffffffe 0x40000000 0x00000003 0x3fca8148 0x00000001 0x3c141884 0x4200f34a
3fca7f60: 0x3c141000 0x0000001f 0x00000003 0x3fca814c 0x3fffffff 0x3ffffffe 0x40000000 0x3fcaacd0
3fca7f80: 0x000000c0 0x3fca8144 0x3fcaaf67 0x40382f12 0x3fffffff 0x3ffffffe 0x3fca8170 0x3fca8130
3fca7fa0: 0x3c14888c 0x3fc9ae5c 0x3fca816c 0x00000000 0x3fca82f4 0x00000002 0x00000069 0x3fca8164
3fca7fc0: 0x0000001e 0x3fcaa540 0x3fca8228 0x42047656 0x403828fa 0x3fca81b0 0x3fca9740 0x00000002
3fca7fe0: 0x00000000 0x3fca81b0 0x0000000b 0x00000000 0x3fca8130 0x00000050 0x00000254 0x3c141000
3fca8000: 0x3c141000 0x3fca7f90 0x3fca9310 0x42025ec4 0x3fca9358 0x3fca9560 0x3c143c54 0x00000000
3fca8020: 0x00000000 0x3fca8310 0x3c143c54 0x00000000 0x00000000 0x3fca8310 0x3c150ae0 0x000004d5
3fca8040: 0x3fcac260 0x3fcab660 0x000004d5 0x42017e6a 0x000005ef 0x3fcab660 0x00000000 0x3fcab660
3fca8060: 0x42015620 0x420155fa 0x3fca80e0 0x3fc97c00 0x3fc6e63c 0x4200c538 0x00000003 0x00000000
3fca8080: 0x00000004 0x00000003 0x00000001 0x3c141000 0x3c141000 0x00000256 0x3fcaa150 0x3fcaaf1f
3fca80a0: 0x3fffffff 0x3c143c90 0x3c1578dc 0x3fca80fc 0x0000000f 0x3fffffff 0x3fcaa090 0x42015204
3fca80c0: 0x3fffffff 0x3ffffffe 0x40000000 0x3fcaa150 0x00000000 0x0000000a 0x00000256 0x3c141000
3fca80e0: 0x3c141000 0x0000000b 0x0000002c 0x00000009 0x3fffffff 0x3ffffffe 0x40000000 0x3c141000
3fca8100: 0x3c141000 0x00000254 0x00000050 0x3fca8130 0x00000000 0x0000000b 0x3fca81b0 0x00000000
3fca8120: 0x00000002 0x3fca9740 0x3fca81b0 0x4200f29a 0x3fca9990 0x3fcaaf64 0x3fca8140 0x0000000b
3fca8140: 0x3fca9c80 0x3c141884 0x3fcaa370 0x00001ac2 0x000000c9 0x000019ba 0x00000001 0x0000119a
3fca8160: 0x3fcaa3a0 0x4c7f70b1 0x3c1420d8 0x3fcaa150 0x000000c0 0x3fcaa2f0 0x3fca81b0 0x4200f1ec
3fca8180: 0x00000001 0x00000010 0x3fcaa150 0x3fcac6cf 0x0000137b 0x3fcaa150 0x3fca9990 0x00000002
3fca81a0: 0x00000000 0x3fca81b0 0x3fca81f0 0x42046f00 0x3fcaa150 0x3c1420d8 0x3fca8388 0x3fca86c0
3fca81c0: 0x403828fa 0x3fca8350 0x3fca9740 0x00000005 0x0000137b 0x00000000 0x00000008 0xd9c0818a
3fca81e0: 0x3c1420d8 0x3fcaa360 0x3fc9ae5c 0x42017ab8 0x3c141000 0x3fca8180 0x0000000f 0x3c1420d8
3fca8200: 0x3c143f83 0x3fcaa020 0x3fc9ae5c 0x42018956 0x00000001 0x3fcaa020 0x3fcaa360 0x3c1420d8
3fca8220: 0x00000000 0x00000000 0x3fca83e8 0x00000000 0x4201894c 0x3fc9ae5c 0x3fcaa360 0x3c1420d8
3fca8240: 0xd9c0818a 0x00000008 0x00000000 0x0000137b 0x3fcac6cf 0x3fcaa150 0x00000254 0x3c141000
3fca8260: 0x3c141000 0x3fca8210 0x00000004 0x0000000c 0x3fcaa154 0x0000000c 0x3fcaa7c8 0x4200ac22
3fca8280: 0x3fffffff 0x0000dc15 0x00000001 0x3fcaa154 0x000012a2 0x00000001 0x3fcaa770 0x40382070

ELF file SHA256: 4e42dac477741998

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807a8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710
MicroPython v1.22.0 on 2023-12-27; LOLIN_C3_MINI with ESP32-C3FH4
Type "help()" for more information.
```

I see in the test.py code the gc.collect() is used, I also tried to gc.collect before the acquire.py code and disabled debug mode with osdebug(None) but did not help..

I can read the raw data and used the same esp for other projects. not sure what to do.

peterhinch, 2024-01-04:
The Apple protocol is quite different from the supported ones and I'm not surprised that the test scripts fail. I don't plan to try to fix this as it's usage outside of the intended scope of these scripts.

Pax-IT, 2024-01-04:
Hi, 

thank you for the reply, but the device rebooting/crashing with unchanged files and trying to test other remotes is not related to my endeavour for getting the Apple Remote working.

The same code on an Wemos S2 mini is working, it seems it's only on the C3. If I'm the only one with this issue on an C3 then ignore my issue but if other C3 owners have the same I think the closing of this issue is incorrect?

peterhinch, 2024-01-04:
Ah, sorry I misunderstood: I assumed it was the Apple sequence which was crashing it.

Unfortunately I don't have a C3 to test. This may be a firmware issue given that the C3 is rather new with a different CPU to other Espressif products. I suggest we see how others get on with these chips.

Pax-IT, 2024-01-04:
Did some more testing on 2 different C3 board:

Working:
```
LOLIN_C3_MINI-20220618-v1.19.1.bin
LOLIN_C3_MINI-20230426-v1.20.0.bin
```
Not working:
```
LOLIN_C3_MINI-20231005-v1.21.0.bin
LOLIN_C3_MINI-20231227-v1.22.0.bin
```
So it seems it starts with 1.21 and is still happening in 1.22

peterhinch, 2024-01-05:
This is almost certainly a firmware issue. A pin interrupt handler is initialising a timer and that call is failing. If I had an ESP32-C3 I would attempt to write a minimal repro and (assuming the error was reproducible) raise an issue.

I'm not sure how to proceed. I could suggest a script to try, if you want to pursue this.

Pax-IT, 2024-01-05:
I happy to help with debugging. I have 2 LOLIN ESP32-C3 boards a v1.0.0 and a v2.1.0

Pax-IT, 2024-01-05:
Updated `__init__.py` line 48 to:
```
self.tim = Timer(0)  # Sofware timer
```
Which looks like it does not break the library code and fix the issue on the C3.

peterhinch, 2024-01-05:
Nice work. I used software timers in the module as I thought these were more portable. Your adaptation uses a hardware timer, which is fine. What troubles me is that the firmware allowed a software timer to be instantiated, with the thing failing on use.

Please could you try this script, which should enable us to raise an issue:
```python
from machine import Timer
import time
tim = Timer(-1)
for _ in range(10):
    print("Trigger")
    tim.init(period=500, mode=Timer.ONE_SHOT, callback=lambda _: print("Timer"))
    time.sleep(1)
```

Pax-IT, 2024-01-05:
I ran your code:
```
>>> from machine import Timer
>>> import time
>>> tim = Timer(-1)
>>> for _ in range(10):
...     print("Trigger")
...     tim.init(period=500, mode=Timer.ONE_SHOT, callback=lambda _: print("Timer"))
...     time.sleep(1)
... 
Trigger
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.
```

peterhinch, 2024-01-05:
I found a previous issue (Oct 2022) and added a comment. We'll see if there's a response.

The above script works on ESP32 and ESP32-S3. If soft timers are not supported, the firmware should throw an exception on line 3 (which could be trapped). Crashing is not good.

peterhinch, 2024-01-07:
I've pushed an update which allows the timer ID to be overridden via a class variable. This should provide ESP32C3 users with a workround. See the docs for an explanation of why a software timer is the default.

Thank you for your work in diagnosing this. I have credited you in the code and docs.

closed 2024-01-07 by peterhinch