peterhinch/micropython_ir #39: Esp32 double transmission

githubniwis, 2024-04-30:
Wanting to control my tv via node-red/mqtt I installed your software on an esp32.

from machine import Pin
from ir_tx.philips import RC6_M0

pin = Pin(15, Pin.OUT,  value = 1)
to_tv = RC6_M0(pin)
to_tv.transmit(0, 9) # 9 is BBC2

When I power up the esp32, everything works fine. However  when I do a restart/stop backend via Thonny, the signal is transmitted twice, so I get 99 instead of 9.
Having spent several evenings tinkering, I have to give up.
Can you help me?

peterhinch, 2024-05-01:
A number of issues in Discussions have been connected to Thonny. It's also difficult to support Thonny problems because it doesn't seem to be possible to produce a session listing. Please could you repeat your testing using the official [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html) tool. If the error occurs, please post a listing of the session and I'll try to figure out what's going on.

githubniwis, 2024-05-01:
Your answer made me realise that  Thonny was the cause of the double transmission, so problem solved.
Everything works just fine.
Thank you for taking the time.

closed 2024-05-01 by githubniwis