peterhinch/micropython_ir #43: Error in playback code in TRANSMITTER.md

bluepole, 2024-12-20:
Small error (editing oversight?) in the playback code in the 'Unsupported Protocols' section where the ESP32 pin is diefined as a 2-tuple rather than an actual Pin. Otherwise good!

peterhinch, 2024-12-20:
Thanks for the report. The error I can see in `RECEIVER.md` section 8 is that
```py
from ir_rx import NEC_16
```
should read
```py
from ir_rx.nec import NEC_16
```
Is this the issue you're reporting?

bluepole, 2024-12-20:
Sorry no - this is the line in the playback code...

`pin = (Pin(23, Pin.OUT, value = 0), Pin(21, Pin.OUT, value = 0))
`

Should be a simple Pin, not a tuple, I think. Anyway, that's what works! Love the project BTW.

peterhinch, 2024-12-21:
Ah, got it! Thanks for the report: now fixed.

closed 2024-12-21 by peterhinch